labscript-suite/labscript #81: Using labscript for continuous acquisition

jondoesntgit, 2021-09-06:
In @philipstarkey's thesis, he outlines the difference between a process control system and a scientific control system, stating that in a scientific control system, you don't usually need to do continuous data measurement.

I run an experiment in my lab where we often run an experiment, and collect data for an indefinite amount of time and save it into an HDF5 file. Using single write multiple read (SWMR), I will often look at how the experiment is progressing, and if there's a problem, stop it and troubleshoot, and if it's running fine, let it continue overnight or over the weekend. Often times, I'm interested in letting an experiment run until I have "enough" data to do an analysis, but it's not always obvious at the beginning how much will be "enough." I'll come in on Monday morning after a weekend, take a peek at the data set while the experiment is still in progress, and at that point decide whether or not to terminate the experiment.

I currently do not use labscript for this experiment, but given the elegance and modularity of the labscript suite, I've considered migrating. However, it appears to me from looking at the codebase and Philip Starkey's thesis that this isn't quite the thing that labscript is built for. Can you confirm/deny?

philipstarkey, 2021-09-17:
Do you need an indefinite length shot with a continuous sample rate? Or would it be OK to acquire data for a few minutes, then have a ~500ms gap in the data, followed by an acquisition of a few minutes, etc.

Basically, if it's a requirement that the day acquisition has no gaps, and a sample rate above 2Hz (which can maybe be pushed to 10Hz), then labscript is probably not suitable. Labscript really only works if acquisitions can be split up, with a small gap between shots where no data acquisition is done (dead time).

Acquisition at 1 MHz with the occasional 500ms gap is fine. Acquisition at 1 MHz for days, with no gaps, probably not fine. 

Can you provide more details of the needed data acquisition?

jondoesntgit, 2021-09-17:
Thanks @philipstarkey, I think that answers the question: best labscript can do is a long shot followed by another long shot. I think we were looking for an indefinite-length shot with continuous sample rate. If the requirement changes, I'll circle back.

philipstarkey, 2021-09-18:
No worries. There is nothing inherent in labscript thatthat limits the length of a shot - it's limited by whatever hardware you use. And technically there is nothing stopping you from writing a custom labscript device for custom hardware that can handle indefinite acquisitions and perform analysis from within BLACS in order to determine when the shot ends....

But it's a very big departure from the design, would require a bunch of custom coding to get working, and doesn't follow the intended design pattern or utilise any of the existing hardware support. So I think it's unlikely the benefits would outweigh the effort!

closed 2021-09-18 by philipstarkey