labscript-suite/labscript #95: It seems that the time sequence cannot be too long.

Moriaty-TT, 2022-08-11:
My initial timing duration is about 25s, and it can run well. But when I tried to cycle my experiment three times, my experiment logic script like this:
```
t = 0
start()
for i in range(3):
     #my experiment. about 25s
stop(t)
```
I‘ve encountered this error：
```
Traceback (most recent call last):
  File "C:\Users\Tao\labscript-suite\userlib\labscriptlib\DFG_MOT_load\DFG_MOT_load_0811.py", line 288, in <module>
    stop(t)
  File "c:\labscript\lib\site-packages\labscript\labscript.py", line 3588, in stop
    generate_code()
  File "c:\labscript\lib\site-packages\labscript\labscript.py", line 3407, in generate_code
    device.generate_code(hdf5_file)
  File "c:\labscript\lib\site-packages\labscript_devices\PrawnBlaster\labscript_devices.py", line 450, in generate_code
    # pulse_program[j]["half_period"] = instruction["half_period"]
OverflowError: Python int too large to convert to C long
Compilation aborted.
```
Is there any way to solve it？Thanks！
@mjasperse @chrisjbillington @philipstarkey

philipstarkey, 2022-08-12:
There was a similar report here: https://groups.google.com/g/labscriptsuite/c/q2sBrLmELn8/m/b4CAIWEcAQAJ

I was never able to reproduce the issue myself locally. Do you have a full example script that reproduces it you would be willing to share?

Moriaty-TT, 2022-08-12:
Thank you Phil!

> a full example script
No problem,but it's a little long.
This is my full example script:
```
from labscript import start,stop,add_time_marker
from labscript.labscript import AnalogOut,DigitalOut
from labscript_devices.NI_DAQmx.labscript_devices import NI_PXIe_6535
from labscript_devices.NI_DAQmx.labscript_devices import NI_PXI_6733
from labscript_devices.NI_DAQmx.labscript_devices import NI_PXIe_6738
from labscript_devices.PrawnBlaster.labscript_devices import PrawnBlaster

#Device define
PrawnBlaster(name='prawn', com_port='COM3',num_pseudoclocks=4)
NI_PXIe_6535(name='Dev1',MAX_name='Dev1',parent_device=prawn.clocklines[0],clock_terminal='/Dev1/PFI4')
NI_PXIe_6535(name='Dev2',MAX_name='Dev2',parent_device=prawn.clocklines[1],clock_terminal='/Dev2/PFI4')
NI_PXI_6733(name='Dev3',MAX_name='Dev3',parent_device=prawn.clocklines[2],clock_terminal='/Dev3/PFI0')
#NI_PXI_6733(name='Dev4',MAX_name='Dev4',parent_device=prawn.clocklines[0],clock_terminal='/Dev4/PFI0')
NI_PXIe_6738(name='Dev5',MAX_name='Dev5',parent_device=prawn.clocklines[3],clock_terminal='/Dev5/PFI0')

#analogout define
#dev5 analogout define
AnalogOut(name='TmpAna',parent_device=Dev5,connection='ao0')
AnalogOut(name='SlowPwr', parent_device=Dev5, connection='ao19')
AnalogOut(name='SlowFreq', parent_device=Dev5, connection='ao2')
AnalogOut(name='laser399_MOTPwr', parent_device=Dev5, connection='ao5')
AnalogOut(name='laser399_MOTFre', parent_device=Dev5, connection='ao6')
AnalogOut(name='MOTCoilI', parent_device=Dev5, connection='ao9')
AnalogOut(name='laser556_MOTFre', parent_device=Dev5, connection='ao7')
AnalogOut(name='laser556_MOTpower', parent_device=Dev5, connection='ao11')
AnalogOut(name='MOTCoilMonitor', parent_device=Dev5, connection='ao10')
AnalogOut(name='Start_coil_I', parent_device=Dev5, connection='ao13')
AnalogOut(name='DecreaseCoilI', parent_device=Dev5, connection='ao14')
AnalogOut(name='Bias_coil_I', parent_device=Dev5, connection='ao15')
AnalogOut(name='MOTBiasX_analog', parent_device=Dev5, connection='ao16')
AnalogOut(name='MOTBiasY_analog', parent_device=Dev5, connection='ao17')
AnalogOut(name='MOTBiasZ_analog', parent_device=Dev5, connection='ao18')
AnalogOut(name='TODTSetpoint', parent_device=Dev5, connection='ao21')
AnalogOut(name='laser399_IMGFre', parent_device=Dev5, connection='ao1')
AnalogOut(name='SODT_setpoint', parent_device=Dev5, connection='ao8')
#dev3 analogout define
AnalogOut(name='TempFastAan', parent_device=Dev3, connection='ao0')
AnalogOut(name='dev3_for_env',parent_device=Dev3,connection='ao1')

#digitalotut define
#dev1 digital out define
DigitalOut(name='TmpTTL', parent_device=Dev1, connection='port0/line0')
DigitalOut(name='AtomShut', parent_device=Dev1, connection='port0/line1')
DigitalOut(name='MOTCoilTTL', parent_device=Dev1, connection='port0/line5')
DigitalOut(name='laser556_MOTTTL', parent_device=Dev1, connection='port0/line6')
DigitalOut(name='SlowShut', parent_device=Dev1, connection='port3/line1')
DigitalOut(name='ZSProShut', parent_device=Dev1, connection='port3/line7')
DigitalOut(name='CCDTri', parent_device=Dev1, connection='port1/line0')
DigitalOut(name='Img_AOM_TTL', parent_device=Dev1, connection='port2/line6')
DigitalOut(name='Img_Shut', parent_device=Dev1, connection='port2/line7')  
DigitalOut(name='Blue_MOT_shut', parent_device=Dev1, connection='port0/line2')
DigitalOut(name='ODT_Tri', parent_device=Dev1, connection='port1/line1')
DigitalOut(name='dev1_for_env', parent_device=Dev1, connection='port2/line5')

#dev2 digital out define
DigitalOut(name='IMG_AOMTTL', parent_device=Dev2, connection='port2/line0')
DigitalOut(name='SODT_TTL', parent_device=Dev2, connection='port1/line2')
DigitalOut(name='SODT_Dumper', parent_device=Dev2, connection='port1/line3')
DigitalOut(name='StartCoil', parent_device=Dev2, connection='port0/line0')
DigitalOut(name='DecreaseCoil', parent_device=Dev2, connection='port0/line1')
DigitalOut(name='BiasCoil', parent_device=Dev2, connection='port0/line2')
DigitalOut(name='MOTBiasX_digital', parent_device=Dev2, connection='port0/line3')
DigitalOut(name='MOTBiasY_digital', parent_device=Dev2, connection='port0/line4')
DigitalOut(name='MOTBiasZ_digital', parent_device=Dev2, connection='port0/line5')
[... 197 lines cut ...]

    t+=1e-4

    #wait CCD
    #digital
    CCDTri.go_low(t)
    Img_AOM_TTL.go_low(t)
    Img_Shut.go_low(t)
    IMG_AOMTTL.go_low(t)

    t+=0.1

    #ImgDark
    #digital
    CCDTri.go_high(t)

    t+=1e-4

    #end
    #digital
    CCDTri.go_low(t)
    t+=2
stop(t)
```
When I only execute a single time sequence, it works well. Once I start the cycle, the above error will occur. I've tried to shorten the sequence time of my single cycle by about 5 second, and when I cycle it three times, it can run. But my actual experiment can not shorten my sequence time.

philipstarkey, 2022-08-12:
So I haven't had chance to try and compile this yet, but looking at it I think I understand what the issue is.

The outputs on `Dev3` aren't used during the shot. And they are the only outputs on PrawnBlaster clockline 2. Which means that labscript attempts to generate a very long single clock tick to cover the majority of the shot length. This exceeds the maximum length of a pulse the PrawnBlaster supports.

It seems like a bit of a deficiency that we don't have a way for pseudoclocks to tell labscript what their maximum clock tick period is. Unfortunately I can't immediately see an easy way to add this into labscript. It has to be done around the `collect_change_times()` part of the generation algorithm. But that happens before ramps are expanded. So it get's a little tricky to calculate what each clock tick length is at that point (I guess it's not impossible though) in order to determine whether to insert additional clock ticks to restrict the period length below the devices maximum.

Anyway...fortunately there is an easy work around. Which is that you just need to use an output on `Dev3` in your loop. You can set it to the same value each time (it doesn't have to change). You just have to force labscript to tick every prawnblaster clockline at least once every 42 seconds. `TempFastAan.constant(t, value=0)` in the `#InitAnalog` section of the loop should do it.

Please let me know if this fixes the issue! We'll keep this open until we have a proper fix in labscript that covers all pseudoclocks.

Moriaty-TT, 2022-08-15:
> You just have to force labscript to tick every prawnblaster clockline at least once every 42 seconds. `TempFastAan.constant(t, value=0)` in the `#InitAnalog` section of the loop should do it.

Thanks Phil！Now it can run successfully！

chrisjbillington, 2026-10-18:
Fixed by labscript-suite/labscript-devices#153, merged 2026-10-18: the PrawnBlaster now raises a LabscriptError naming the clockline and the maximum interval between ticks, with the workaround.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-10-18 by chrisjbillington